screetsec/Brutal #9: Required New Version Of BRUTAL V.2 :)

MeanMachine123, 2018-01-07:
I need a new version of BRUTAL with new Features & Functions :)

root09005, 2018-06-26:
Me to please

root09005, 2018-07-01:
can i use the script aslo for an rubber ducky

watbulb, 2019-04-12:
@MeanMachine123 Free software gets free support. Would you just build a new version of something because someone just "asked you to"? These guys aren't getting paid for this, keep that in mind.

Fork it, propose new ideas, contribute.

MeanMachine123, 2023-02-03:
> @MeanMachine123 Free software gets free support. Would you just build a new version of something because someone just "asked you to"? These guys aren't getting paid for this, keep that in mind.
> 
> Fork it, propose new ideas, contribute.

Thats Y secreetsec is my favorite. I am in love with his tools. So I want more to use 🥇